DIGIMAD-4/main #10: Interboard communicatieprotocol ontwerpen.

Villadelfia, 2020-11-26:


Villadelfia, 2020-11-26:
Beschrijving hardwareopstelling:

- Elke beacon module heeft een "A" hardware UART poort, en een "B" software serial UART poort.
- De Raspberry Pi is direct verbonden met de "A" poort van de master beacon (0).
- De "B" poort van de master beacon is verbonden met de "A" poort van de eerste slave beacon (1).
- De "B" poort van de eerste slave beacon is verbonden met de "A" poort van de tweede slave beacon (2).
- De "B" poort van de tweede slave beacon is verbonden met de "A" poort van de derde slave beacon (3).
- Op de tracker PCBs is de ID bepaald door middel van jumpers.

Beschrijving firmware opstelling:

- Elke beacon module draait exact dezelfde firmware. Een state machine met de volgende staten: _IDLE_, PING, WAIT
- Elke tracker draait dezelfde firmware. Deze firmware is een zeer strakke lus die constant luistert voor zijn ID. Zodra deze gehoord is stuurt hij deze terug.

Gedetailleerd werkingsidee:

1. Algoritme op Raspberry Pi bepaalt volgende tracker ID om te meten. Stuurt vervolgens twee bytes over UART naar beacon 0: (uint8)trackerID, (uint8)beaconID = 0.
2. Beacon 0, in IDLE staat, ontvangt het commando, en stuurt (uint8)0xAA terug. Indien verzender deze confirmatie niet ontvangt, wordt her commando herhaald. Beacon 0 slaat lokaal trackerID en beaconID op en gaat naar de PING staat.
3. Beacon 0 stuurt de trackerID in een 8 bit radiopakket, en wacht op respons met herhaling na 1 seconde. Signaalsterkte wordt opgeslagen, en tijd in klokcycli sinds verzenden ping wordt opgeslagen. Deze meting wordt drie maal snel na elkaar herhaald en het gemiddelde van de drie delays en signaalsterktes wordt berekend en opgeslagen.
4. Beacon 0 verhoogt beaconID met 1 en stuurt vervolgende twee bytes naar beacon 1: (uint8)trackerID, (uint8)beaconID. Bij ontvangst van confirmatie (0xAA) van beacon 1, gaat beacon 0 naar de WAIT staat.
5. Herhaal stappen 2, 3, en 4 voor beacon 1.
6. Herhaal stappen 2, 3, en 4 voor beacon 2.
7. Herhaal stappen 2 en 3 voor beacon 3.
8. Omdat beaconID nu 3 is, weet beacon 3 dat het de laatste is, en gaat deze naar de WAIT staat. Hij stuurt nu de volgende data terug naar beacon 2: (uint8)beaconID, (uint8)RSSI, (uint32)delay, (uint16)CRC16. Bij het ontvangen van 0xAA van beacon 2 gaat beacon 3 naar de IDLE staat, bij elke andere ontvangst wordt de data opnieuw verzonden.
9. Beacon 2 ontvangt de data van beacon 3, en checkt de CRC. Bij een correcte CRC stuurt hij 0xAA terug, anders 0x55. Hij stript nu de CRC16 van de data, en voegt zijn eigen data er aan toe om verder te sturen naar beacon 1: (6 bytes)data, (uint8)beaconID, (uint8)RSSI, (uint32)delay, (uint16)CRC16. Bij het ontvangen van 0xAA van beacon 1 gaat beacon 2 naar de IDLE staat, bij elke andere ontvangst wordt de data opnieuw verzonden.
10. Beacon 1 ontvangt de data van beacon 2, en checkt de CRC. Bij een correcte CRC stuurt hij 0xAA terug, anders 0x55. Hij stript nu de CRC16 van de data, en voegt zijn eigen data er aan toe om verder te sturen naar beacon 0: (12 bytes)data, (uint8)beaconID, (uint8)RSSI, (uint32)delay, (uint16)CRC16. Bij het ontvangen van 0xAA van beacon 0 gaat beacon 1 naar de IDLE staat, bij elke andere ontvangst wordt de data opnieuw verzonden.
11. Beacon 0 ontvangt de data van beacon 1, en checkt de CRC. Bij een correcte CRC stuurt hij 0xAA terug, anders 0x55. Hij stript nu de CRC16 van de data, en voegt zijn eigen data er aan toe om verder te sturen naar Raspberry Pi: (18 bytes)data, (uint8)beaconID, (uint8)RSSI, (uint32)delay, (uint16)CRC16. Bij het ontvangen van 0xAA van Raspberry Pi gaat beacon 1 naar de IDLE staat, bij elke andere ontvangst wordt de data opnieuw verzonden.
12. Raspberry Pi ontvangt de uiteindelijke data van de meting, 24 bytes + 2 bytes CRC. Bij een correcte CRC stuurt hij 0xAA terug, anders 0x55. Vervolgens worden de metingen omgezet naar X, Y, Z coördinaten en doorgezonden naar het IT team. De volledige ontvangen data ziet er nu zo uit:

```
(1 byte)3
(1 byte)RSSI beacon 3
(4 bytes)Delay beacon 3
(1 byte)2
(1 byte)RSSI beacon 2
(4 bytes)Delay beacon 2
(1 byte)1
(1 byte)RSSI beacon 1
(4 bytes)Delay beacon 1
(1 byte)0
(1 byte)RSSI beacon 0
(4 bytes)Delay beacon 0
```

closed 2020-12-03 by Villadelfia

Villadelfia, 2020-12-03:
Update aan tracker code: De RF96 ondersteunt message ID, FROM en TO header filtering. Dus de chipset wordt in de plaats zo geconfigureerd dat enkel de messages bestemd voor een tracker daar aan komen, idem voor de beacons. Er moet dus NIET gecheckt worden op de juiste ID bij het aankomen, maar de bestemmeling moet correct ingesteld worden bij verzending.

20007082, 2020-12-08:
Kan er mij rechten gegeven worden zodat ik ook issues kan verplaatsen in het Kanban bord. Zoals dit besproken is afgelopen week?

Villadelfia, 2020-12-08:
Nieuwe invite gestuurd.
https://i.imgur.com/5lH7VsY.png